isaac-sim/OmniIsaacGymEnvs #121: Unable to install requirements

Cdfghglz, 2023-12-15:
The local install fails with

```bash
cdf@xps15:~/ws/pr_omni_isaac_gym_envs$  (main)PYTHON_PATH -m pip install -e .
Obtaining file:///home/cdf/ws/pr_omni_isaac_gym_envs
Requirement already satisfied: numpy==1.23.5 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/exts/omni.kit.pip_archive/pip_prebundle (from omniisaacgymenvs==2023.1.0b0) (1.23.5)
Requirement already satisfied: protobuf==3.20.2 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from omniisaacgymenvs==2023.1.0b0) (3.20.2)
Requirement already satisfied: omegaconf==2.3.0 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from omniisaacgymenvs==2023.1.0b0) (2.3.0)
Requirement already satisfied: hydra-core==1.3.2 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from omniisaacgymenvs==2023.1.0b0) (1.3.2)
Requirement already satisfied: urllib3==1.26.16 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/exts/omni.kit.pip_archive/pip_prebundle (from omniisaacgymenvs==2023.1.0b0) (1.26.16)
Requirement already satisfied: rl-games==1.6.1 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from omniisaacgymenvs==2023.1.0b0) (1.6.1)
Requirement already satisfied: packaging in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/exts/omni.isaac.core_archive/pip_prebundle (from hydra-core==1.3.2->omniisaacgymenvs==2023.1.0b0) (23.0)
Requirement already satisfied: antlr4-python3-runtime==4.9.* in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from hydra-core==1.3.2->omniisaacgymenvs==2023.1.0b0) (4.9.3)
Requirement already satisfied: PyYAML>=5.1.0 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/exts/omni.pip.compute/pip_prebundle (from omegaconf==2.3.0->omniisaacgymenvs==2023.1.0b0) (6.0.1)
Requirement already satisfied: opencv-python<5.0.0,>=4.5.5 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (4.8.1.78)
Requirement already satisfied: gym[classic-control]<0.24.0,>=0.23.0 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.23.1)
Requirement already satisfied: tensorboard<3.0.0,>=2.8.0 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2.15.1)
Requirement already satisfied: wandb<0.13.0,>=0.12.11 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.12.21)
Requirement already satisfied: setproctitle<2.0.0,>=1.2.2 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (1.3.3)
Requirement already satisfied: tensorboardX<3.0,>=2.5 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2.6.2.2)
Requirement already satisfied: psutil<6.0.0,>=5.9.0 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (5.9.6)
WARNING: gym 0.23.1 does not provide the extra 'classic-control'
Requirement already satisfied: cloudpickle>=1.2.0 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from gym[classic-control]<0.24.0,>=0.23.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (3.0.0)
Requirement already satisfied: gym-notices>=0.0.4 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from gym[classic-control]<0.24.0,>=0.23.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.0.8)
Requirement already satisfied: absl-py>=0.4 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2.0.0)
Requirement already satisfied: requests<3,>=2.21.0 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/exts/omni.pip.cloud/pip_prebundle (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2.31.0)
Requirement already satisfied: google-auth-oauthlib<2,>=0.5 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (1.2.0)
Requirement already satisfied: setuptools>=41.0.0 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/exts/omni.kit.pip_archive/pip_prebundle (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (68.0.0)
Requirement already satisfied: tensorboard-data-server<0.8.0,>=0.7.0 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.7.2)
Requirement already satisfied: markdown>=2.6.8 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (3.5.1)
Requirement already satisfied: google-auth<3,>=1.6.3 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2.25.2)
Requirement already satisfied: six>1.9 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/exts/omni.kit.pip_archive/pip_prebundle (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (1.16.0)
Requirement already satisfied: grpcio>=1.48.2 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (1.60.0)
Requirement already satisfied: werkzeug>=1.0.1 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (3.0.1)
Requirement already satisfied: rsa<5,>=3.1.4 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from google-auth<3,>=1.6.3->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (4.9)
Requirement already satisfied: pyasn1-modules>=0.2.1 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from google-auth<3,>=1.6.3->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.3.0)
Requirement already satisfied: cachetools<6.0,>=2.0.0 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from google-auth<3,>=1.6.3->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (5.3.2)
Requirement already satisfied: requests-oauthlib>=0.7.0 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/exts/omni.pip.cloud/pip_prebundle (from google-auth-oauthlib<2,>=0.5->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (1.3.1)
Requirement already satisfied: pyasn1<0.6.0,>=0.4.6 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages (from pyasn1-modules>=0.2.1->google-auth<3,>=1.6.3->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.5.1)
Requirement already satisfied: certifi>=2017.4.17 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/exts/omni.kit.pip_archive/pip_prebundle (from requests<3,>=2.21.0->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2023.5.7)
Requirement already satisfied: charset-normalizer<4,>=2 in /home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/exts/omni.kit.pip_archive/pip_prebundle (from requests<3,>=2.21.0->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2.1.1)
[... cut ...] e('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps Check the logs for full command output.
WARNING: You are using pip version 21.2.1+nv1; however, version 23.3.1 is available.
You should consider upgrading via the '/home/cdf/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/python/bin/python3 -m pip install --upgrade pip' command.                                                  
There was an error running python
```

Any help appreciated

kellyguo11, 2024-01-03:
Hi there, please make sure you are running the `pip install -e .` command from within the OmniIsaacGymEnvs folder.

maxencefaldor, 2024-01-15:
I have the same issue. Any update?

Cdfghglz, 2024-01-16:
I personally moved to Docker for now, so not resolved for me.