isaac-sim/OmniIsaacGymEnvs #123: Bug in Anymal terrain task

as-wanfang, 2023-12-21:
It seems there is a typo error in anymal_terrain.py, which specifies the commanded rotation angular velocity around z-axis

line 275: self.commands[env_ids, 3] = torch_rand_float(self.command_yaw_range[0], self.command_yaw_range[1], (len(env_ids), 1), device=self.device).squeeze()

The index should be 2 instead of 3.

line 341: self.commands[:, 2] = ...
The index 2 might also be wrong?

kellyguo11, 2024-01-03:
Hi there, thanks for reporting this. We will look into it.

as-wanfang, 2024-01-06:
After further reading the codes, I found this might not be a bug but a typo error in the task config file line 37: yaw: [-3.14, 3.14]    # min max [rad/s]. If the code in tasks/anymal_terrain.py is right, then the range in config file is the range of heading instead range of angular velocity.

However, in demos/anymal_terrain.py, 

    def set_up_keyboard(self):
        self._input = carb.input.acquire_input_interface()
        self._keyboard = omni.appwindow.get_default_app_window().get_keyboard()
        self._sub_keyboard = self._input.subscribe_to_keyboard_events(self._keyboard, self._on_keyboard_event)
        T = 1
        R = 1
        self._key_to_control = {
            "UP": [T, 0.0, 0.0, 0.0],
            "DOWN": [-T, 0.0, 0.0, 0.0],
            "LEFT": [0.0, T, 0.0, 0.0],
            "RIGHT": [0.0, -T, 0.0, 0.0],
            "Z": [0.0, 0.0, R, 0.0],
            "X": [0.0, 0.0, -R, 0.0],
        }

the rotation value should be assigned to self.commands[:, 3], which is later used to calculate self.commands[:, 2].