isaac-sim/OmniIsaacGymEnvs #132: Something wrong in ArticulationView of Isaac Sim2023.1.1

MaoZhengquan, 2024-01-17:
I just imported the robot from my urdf file.When I train by rlgames_train.py,something is probably wrong.
This is running under Ubuntu22.04,the version of isaac sim is 2023.1.1
2024-01-17 02:45:06 [27,424ms] [Error] [omni.physx.tensors.plugin] Pattern '/World/envs/*/humrob_v3' did not match any articulations

I can run it normally under Windows,the version of isaac sim is 2023.1.0-hotfix.1, the usd file is the same.
So l try to uninstall isaac sim 2023.1.1 and install the previous version,but it had other error messages
2024-01-17 00:54:08 [6,958ms] [Error] [omni.kit.app._impl] [py stderr]: AttributeError: 'VecEnvRLGames' object has no attribute 'world'

Has anyone else encountered similar issues?

makolon, 2024-02-01:
Hi, @MaoZhengquan !
I am facing the same problem. Have you found a solution?

MaoZhengquan, 2024-02-06:
@makolon I downgraded the version to 2023.1.0-hotfix.1 and replaced the OIGE file downloaded before, and then it can run normally.Hope it helps you

MaoZhengquan, 2024-02-06:
@eferreirafilho Thank you for your advice, I hope it helps others solve this problem!