isaac-sim/OmniIsaacGymEnvs #144: ModuleNotFoundError: No module named 'gymnasium'

overrize, 2024-03-06:
When I run the example rlgame_train.py,it shows ModuleNotFoundError: No module named 'gymnasium' even in the conda enviroments.
``Warning: running in conda env, please deactivate before executing this script
If conda is desired please source setup_conda_env.sh in your python 3.10 conda env and run python normally
Traceback (most recent call last):
  File "/home/xx/Documents/isaacsim/OmniIsaacGymEnvs/omniisaacgymenvs/scripts/rlgames_train.py", line 37, in <module>
    from omniisaacgymenvs.envs.vec_env_rlgames import VecEnvRLGames
  File "/home/xxx/.local/share/ov/pkg/isaac_sim-2023.1.1/OmniIsaacGymEnvs/omniisaacgymenvs/envs/vec_env_rlgames.py", line 34, in <module>
    from omni.isaac.gym.vec_env import VecEnvBase
  File "/home/xxx/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.gym/omni/isaac/gym/vec_env/__init__.py", line 9, in <module>
    from omni.isaac.gym.vec_env.vec_env_base import VecEnvBase
  File "/home/xxx/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.gym/omni/isaac/gym/vec_env/vec_env_base.py", line 14, in <module>
    import gymnasium as gym
ModuleNotFoundError: No module named 'gymnasium'
There was an error running python

overrize, 2024-03-06:
I wander what case this problem

MaltheT, 2024-03-11:
Hmm, I have the same problem

aswinaus, 2024-11-16:
This is what worked for me
pip show gym
#The command above will output the **Location** of gym
import sys
#Please make sure the Location from the above is mentioned below within parenthesis
sys.path.append('C:\\Users\\<profilename>\\anaconda3\\Lib\\site-packages')
import gym