isaac-sim/OmniIsaacGymEnvs #150: Not able to drag in assets from Isaac Assets after the startup hang fix

SuomiKP31, 2024-03-17:
After commit **7e80e14f45fff853342f43e4a25713c18d3aa076**, using this extension seems to cause Isaac Assets to not work correctly.
I noticed that the version of Isaac Sim has been changed to 2023.1.1a while I'm on the release version 2023.1.1. I suppose the issue will be fixed in next major release?

I got this error message:
```
2024-03-17 23:17:25 [847,470ms] [Warning] [omni.usd._impl.layer_legacy] omni.usd.Layers is DEPRECATED. Please update to omni.kit.usd.layers for new layers interface.
2024-03-17 23:17:25 [847,488ms] [Error] [omni.kit.commands.command] Failed to execute a command: CreatePayloadCommand.
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.kit.viewport.window/omni/kit/viewport/window/dragdrop/handler.py", line 177, in <lambda>
    viewport_api.request_query(mouse, lambda *args: self.__query_complete(is_drop, *args),
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.kit.viewport.window/omni/kit/viewport/window/dragdrop/handler.py", line 147, in __query_complete
    instance.dropped(drop_data)
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.kit.viewport.window/omni/kit/viewport/window/dragdrop/usd_file_drop_delegate.py", line 151, in dropped
    usd_prim_path = self.add_usd_drop_marker(drop_data, self.__world_space_pos)
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.kit.viewport.window/omni/kit/viewport/window/dragdrop/usd_file_drop_delegate.py", line 174, in add_usd_drop_marker
    new_prim_path, edit_context, relative_url = self.add_reference_to_stage(usd_context, stage, url)
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.kit.viewport.window/omni/kit/viewport/window/dragdrop/scene_drop_delegate.py", line 180, in add_reference_to_stage
    omni.kit.commands.execute(cmd_name, usd_context=usd_context, path_to=new_prim_path, asset_path=url, instanceable=instanceable)
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.kit.commands/omni/kit/commands/command.py", line 463, in execute
    result = omni.kit.undo.execute(command, name, kwargs)
[...skipped...]
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.kit.commands/omni/kit/undo/undo.py", line 77, in execute
    result = _execute(command, name, level, history_key)
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.kit.commands/omni/kit/undo/undo.py", line 458, in _execute
    raise error
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.kit.commands/omni/kit/undo/undo.py", line 419, in _execute
    result = command.do()
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.usd/omni/usd/commands/usd_commands.py", line 5026, in do
    prim_spec.SetInstanceable(

 <class 'AttributeError'> 'PrimSpec' object has no attribute 'SetInstanceable'
2024-03-17 23:17:25 [847,489ms] [Error] [omni.usd.commands.usd_commands] Invalid prim path to transform
2024-03-17 23:17:25 [847,491ms] [Error] [omni.kit.viewport.window.dragdrop.handler] Traceback:
Traceback (most recent call last):
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.kit.viewport.window/omni/kit/viewport/window/dragdrop/handler.py", line 147, in __query_complete
    instance.dropped(drop_data)
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.kit.viewport.window/omni/kit/viewport/window/dragdrop/usd_file_drop_delegate.py", line 153, in dropped
    omni.kit.commands.create('TransformPrimSRTCommand', path=usd_prim_path, new_translation=self.__world_space_pos).do()
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.usd/omni/usd/commands/usd_commands.py", line 2660, in do
    self._set_transform_srt(
  File "c:\users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/kernel/py\carb\profiler\__init__.py", line 83, in wrapper
    r = f(*args, **kwds)
  File "c:/users/fangz/appdata/local/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.usd/omni/usd/commands/usd_commands.py", line 2358, in _set_transform_srt
    scale = Gf.Vec3d(scale_in)
Boost.Python.ArgumentError: Python argument types in
    Vec3d.__init__(Vec3d, NoneType)
did not match C++ signature:
    __init__(struct _object * __ptr64, double, double, double)
    __init__(struct _object * __ptr64, double)
    __init__(struct _object * __ptr64, class pxrInternal_v0_22__pxrReserved__::GfVec3d)
    __init__(struct _object * __ptr64, class pxrInternal_v0_22__pxrReserved__::GfVec3i)
    __init__(struct _object * __ptr64, class pxrInternal_v0_22__pxrReserved__::GfVec3h)
    __init__(struct _object * __ptr64, class pxrInternal_v0_22__pxrReserved__::GfVec3f)
    __init__(class boost::python::api::object)

2024-03-17 23:17:25 [847,495ms] [Warning] [omni.anim.skelJoint.plugin] Prim /World/SMB_JACK_TO_FAKRA_PLUG_ADAPTER__JFb is invalid when skeljoint processing prim resync
```
Restarting Isaac Sim does not resolve this issue, even if the OIGE plugin is not loaded.

Loading directly from content server still works, for example:
```
def create_test_env():
    if World.instance():
        World.instance().clear_instance()
    world=World()
    world.scene.add_default_ground_plane(z_position=-1.0)
    # Load "Ant" from Isaac Assets
    # usd_path = "http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2023.1.1/Isaac/Robots/Ant/ant.usd"
    # prim_path = "/World/ant"
    usd_path = "http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2023.1.1/Isaac/Robots/Humanoid/humanoid.usd"
    prim_path = "/World/humanoid"
    stage_utils.add_reference_to_stage(usd_path, prim_path)
```
This will load one humanoid into the scene, however still cause the entire app to freeze for about 2 minutes, like before the aforementioned commit.
What is the root cause of this content server related hang?

linhezi, 2024-05-29:
when I drag a sensor into stage , I also meet this issue , have you solve this problem? thx

SuomiKP31, 2024-06-02:
> when I drag a sensor into stage , I also meet this issue , have you solve this problem? thx

Basically any playload will break. No solutions yet, but you can use the script editor to load the prim manually, as a workaround.
I've had the code snippet in the issue but I think things still need to be fixed on NVidia side.